antdroid-hexapod/antdroid #16: catkin_make doesn't make antdroid_msgs/*.h files

stuartm2, 2016-12-16:
Following instructions on the wiki (on RPi2 + Ubuntu 14.04 + Indigo). During 'catkin_make', the antdroid_msgs/*.h files don't seem to be created although the whole process completes and reports success.  When subsequently running 'catkin_make antdroid_antfirm_firmware_upload', it fails with:

```
In file included from /home/ubuntu/catkin_ws/src/antdroid/antdroid_antfirm/firmware/controlRos.cpp:61:0:
/home/ubuntu/catkin_ws/src/antdroid/antdroid_antfirm/firmware/controlRos.h:30:32: fatal error: antdroid_msgs/Walk.h: No such file or directory
 #include <antdroid_msgs/Walk.h>
                                ^
compilation terminated.
```

stuartm2, 2016-12-16:
Here's the full output of catkin_make:

```
ubuntu@ubuntu:~/catkin_ws$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "cmake /home/ubuntu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/ubuntu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ubuntu/catkin_ws/install -G Unix Makefiles" in "/home/ubuntu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - antdroid (metapackage)
-- ~~  - antdroid_msgs
-- ~~  - antdroid_control_interpreter
-- ~~  - antdroid_bringup
-- ~~  - antdroid_cam_control
-- ~~  - antdroid_teleop
-- ~~  - antdroid_antfirm
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'antdroid'
-- ==> add_subdirectory(antdroid/antdroid)
-- +++ processing catkin package: 'antdroid_msgs'
-- ==> add_subdirectory(antdroid/antdroid_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- antdroid_msgs: 10 messages, 0 services
-- +++ processing catkin package: 'antdroid_control_interpreter'
-- ==> add_subdirectory(antdroid/antdroid_control_interpreter)
-- +++ processing catkin package: 'antdroid_bringup'
-- ==> add_subdirectory(antdroid/antdroid_bringup)
-- +++ processing catkin package: 'antdroid_cam_control'
-- ==> add_subdirectory(antdroid/antdroid_cam_control)
-- +++ processing catkin package: 'antdroid_teleop'
-- ==> add_subdirectory(antdroid/antdroid_teleop)
-- +++ processing catkin package: 'antdroid_antfirm'
-- ==> add_subdirectory(antdroid/antdroid_antfirm)
CMake Warning at /home/ubuntu/catkin_ws/build/antdroid/antdroid_antfirm/cmake/antdroid_antfirm-genmsg.cmake:3 (message):
  Invoking generate_messages() without having added any message or service
  file before.

  You should either add add_message_files() and/or add_service_files() calls
  or remove the invocation of generate_messages().
Call Stack (most recent call first):
  /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  antdroid/antdroid_antfirm/CMakeLists.txt:24 (generate_messages)

-- antdroid_antfirm: 0 messages, 0 services
-- Using rosserial_arduino/make_libraries.py to make rosserial client library.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/ubuntu/catkin_ws/build"
####
Scanning dependencies of target _antdroid_msgs_generate_messages_check_deps_Walk
[... 58 lines cut ...]
[ 72%] Built target antdroid_msgs_generate_messages_py
[ 75%] Generating C++ code from antdroid_msgs/Speed.msg
[ 77%] Generating C++ code from antdroid_msgs/Gait.msg
[ 80%] Generating C++ code from antdroid_msgs/MoveLeg.msg
[ 83%] Generating C++ code from antdroid_msgs/Rotate.msg
[ 86%] Generating C++ code from antdroid_msgs/Log.msg
[ 88%] Generating C++ code from antdroid_msgs/Walk.msg
[ 91%] Generating C++ code from antdroid_msgs/Foot.msg
[ 91%] Built target antdroid_msgs_generate_messages_cpp
Scanning dependencies of target antdroid_msgs_generate_messages
Scanning dependencies of target antdroid_msgs_gencpp
[ 91%] [ 91%] Built target antdroid_msgs_gencpp
Built target antdroid_msgs_generate_messages
Scanning dependencies of target antdroid_teleop_joy
Scanning dependencies of target control_interpreter
[ 94%] [ 97%] Building CXX object antdroid/antdroid_teleop/CMakeFiles/antdroid_teleop_joy.dir/src/antdroid_joy.cpp.o
Building CXX object antdroid/antdroid_control_interpreter/src/CMakeFiles/control_interpreter.dir/control_interpreter_core.cpp.o
[100%] Building CXX object antdroid/antdroid_control_interpreter/src/CMakeFiles/control_interpreter.dir/main.cpp.o
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/antdroid_cam_control/cam_control
[100%] Built target cam_control
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/antdroid_teleop/antdroid_teleop_joy
[100%] Built target antdroid_teleop_joy
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/antdroid_control_interpreter/control_interpreter
[100%] Built target control_interpreter
```

stuartm2, 2016-12-16:
With the following diff:

```
diff --git a/antdroid_antfirm/CMakeLists.txt b/antdroid_antfirm/CMakeLists.txt
index d427074..371d45f 100644
--- a/antdroid_antfirm/CMakeLists.txt
+++ b/antdroid_antfirm/CMakeLists.txt
@@ -9,15 +9,16 @@ find_package(catkin REQUIRED COMPONENTS
 )
 
 
-#add_message_files(
-#    FILES
-#    Balance.msg
-#    Foot.msg
-#    Height.msg
-#    Rotate.msg
-#    Speed.msg
-#    Walk.msg
-#)
+add_message_files(
+  DIRECTORY   ../antdroid_msgs/msg
+  FILES
+    Balance.msg
+    Foot.msg
+    Height.msg
+    Rotate.msg
+    Speed.msg
+    Walk.msg
+)
 
 
 ## Generate added messages and services with any dependencies listed here
@@ -42,4 +43,4 @@ rosserial_configure_client(
     )
  
 rosserial_add_client_target(firmware ALL)
-rosserial_add_client_target(firmware upload)
\ No newline at end of file
+rosserial_add_client_target(firmware upload)
```

The message header files compile but I then hit issues with C imports:

```
[ 57%] Building CXX object CMakeFiles/mega2560_firmware.dir/controlRos.cpp.obj
In file included from /home/ubuntu/catkin_ws/src/antdroid/antdroid_antfirm/firmware/controlRos.h:30:0,
                 from /home/ubuntu/catkin_ws/src/antdroid/antdroid_antfirm/firmware/controlRos.cpp:61:
/home/ubuntu/catkin_ws/devel/include/antdroid_msgs/Walk.h:9:18: fatal error: string: No such file or directory
 #include <string>
                  ^
compilation terminated.
```

pando85, 2016-12-26:
I could reproduce the problem using docker.

Create `Dockerfile`:
```Dockerfile
FROM pando85/armhf-ros-indigo

RUN apt-get update && apt-get install --no-install-recommends -y \
    ros-indigo-ros-comm ros-indigo-diagnostics ros-indigo-joystick-drivers ros-indigo-rosserial ros-indigo-rosserial-server ros-indigo-rosserial-arduino ros-indigo-robot-upstart ros-indigo-bond-core ros-indigo-dynamic-reconfigure ros-indigo-nodelet-core ros-indigo-class-loader ros-indigo-image-common ros-indigo-vision-opencv ros-indigo-image-transport-plugins \
    && rm -rf /var/lib/apt/lists/*

RUN apt-get update && apt-get install --no-install-recommends -y \
    git \
    arduino \
    && rm -rf /var/lib/apt/lists/*

RUN apt-get update && apt-get install --no-install-recommends -y \
    build-essential \
    && rm -rf /var/lib/apt/lists/*

RUN mkdir -p /catkin_ws/src && \
    /bin/bash -c '. /opt/ros/indigo/setup.bash; catkin_init_workspace /catkin_ws/src'

RUN /bin/bash -c '. /opt/ros/indigo/setup.bash; cd /catkin_ws; catkin_make'

RUN cd /catkin_ws/src && \
    git clone https://github.com/antdroid-hexapod/antdroid.git 

RUN /bin/bash -c '. /opt/ros/indigo/setup.bash; cd /catkin_ws; catkin_make'

```

``` bash
docker build  . -t antdroid 
docker run antdroid /bin/bash -c '. /opt/ros/indigo/setup.bash; cd /catkin_ws; catkin_make antdroid_antfirm_firmware_upload'
```

pando85, 2016-12-26:
I find a workarround. If you remove your build directory and recompile it, its works perfectly.

You can try in docker:
```bash
$ docker run antdroid /bin/bash -c '. /opt/ros/indigo/setup.bash; cd /catkin_ws; catkin_make antdroid_antfirm_firmware_upload | rm -rf build && catkin_make antdroid_antfirm_firmware_upload'
```

Or in raspberry pi:
```bash
$ catkin_make antdroid_antfirm_firmware_upload | rm -rf build && catkin_make antdroid_antfirm_firmware_upload'
```

stuartm2, 2017-01-31:
Thanks, I can confirm that workaround fixed the upload issue for me.

hhuaibo, 2018-03-08:
-- Could not find the required component 'rosserial_arduino'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "rosserial_arduino"
  with any of the following names:

    rosserial_arduinoConfig.cmake
    rosserial_arduino-config.cmake

  Add the installation prefix of "rosserial_arduino" to CMAKE_PREFIX_PATH or
  set "rosserial_arduino_DIR" to a directory containing one of the above
  files.  If "rosserial_arduino" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  antdroid/antdroid_antfirm/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/charles/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/charles/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

hhuaibo, 2018-03-08:
root@charles-pc:/catkin_ws# docker build  . -t antdroid 
docker: "build" requires 1 argument. See 'docker build --help'.
root@charles-pc:/catkin_ws#  docker run antdroid /bin/bash -c '. /opt/ros/indigo/setup.bash; cd /catkin_ws; catkin_make antdroid_antfirm_firmware_upload || rm -rf build && catkin_make antdroid_antfirm_firmware_upload'
Unable to find image 'antdroid:latest' locally
Pulling repository antdroid
FATA[0006] Error: image library/antdroid:latest not found