antdroid-hexapod/antdroid #18: Hexapod ROS control and calibration

Vis-Existence, 2018-06-05:
Hi I have been working for a few months building this and attempting to understand the ROS software.

I am using a Raspberry Pi 2 B running Kinetic Ubuntu 16.04 from this source: http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

I have currently setup my Raspberry Pi connected to a Arduino Mega 2650 that is all installed into the Hexapod and installed ROS and have managed to get the "antfirm.launch" running correctly as it states its "connected and ready to receive commands"

I have also setup another RPI 2B with the same setup running a desktop install and I can launch and run roscore. 

From here I am getting stuck on how to actually communicate with he Hexapod from my "Master/Talker" system to run basic calibration and get it to move apart from the servo twitch that happens when I run "Antfirm.launch" 

Does anyone have any guides or list of basic commands I need to run in new terminal windows to get this thing to start moving and gain control.

Cheers

Vis

Marcelomm103, 2018-07-10:
Hi Vis

Did u got a solution for the calibration? I’m having problems with that as well.

I’m running ROS kinect on a raspberry pi 3 with ubuntu 16.04, using ssh. Everything compiled without errors.

But when i start antdroid service, all servos just keep moving a little, i try to send calibrate msgs, and keep getting just this msg: Publishing and latching msg. Press Ctrl C to terminate. And nothing ever changes. If i run rostopic list, there isnt any calibrate topic.

Thanks! 
Marcelo

Vis-Existence, 2018-07-10:
I'm in a very similar position myself, and its good to hear I'm not the only one. 

I have been doing a lot or reading about ROS (as this is my first time with anything like this.) I found the guide was good to get into ROS and learn the basics. But once you have understood how to build this (in ROS) there isn't any guides to state how to actually use it. The internet has outdated info or just videos of people who have theirs working. And anything to do with ROS is too in depth or unrelated for my level.

I have two RPi2B setup, one as a "Talker" Master URI running Ubuntu 16.4 on Kinetic which I SSH into via my Win 10 PC. I have that running ROSCORE. Then the second with the same Ubuntu build but set as the "Listener" both currently via Ethernet (as I cant work out the wifi yet) Attached via USB serial I have a Arduino Mega 2560 clone with a Sensor Shield 2.0 to breakout the 18 servos. 

I have been having problems with the "**catkin_make antdroid_antfirm_firmware_upload**" command. As it didn't recognise the ttyUSB serial and was trying to use ttyACM. I corrected a few line of code and got it to write to the Mega. I fried a few Mega's working out how not to overload the power circuit, as I didn't want to drain my Li-Po and was running off 5v mains. Accidentally forgot to shutdown and connect everything via Li-Po and caused a surge....  

The most I have been able to produce myself when running "Antfirm.Launch" was a servo twitch. Most legs just spasm and lock up and a couple may twitch. 

Check out my Crafting blog as I post my progress there (https://www.facebook.com/VisCrafting/) Also let me if you have anything good resources as I could do with some pointers myself! :)

Sorry for the long post but its good to find someone at the same stage as me. 

Cheers Vis