antdroid-hexapod/antdroid #19: Various issues

smiffy6969, 2021-02-15:
Hi, thanks for this project and the time put into it, I decided to try and and have a go, see if I could get this up and running but im meeting with many issues.... where to start....

I suppose the biggest issue is mainly due to the age of the project and the repos to start with. 

I initially tried setting this up on raspian... which is now raspberry pi os. But I had issues due to the age of some of the repo's so I went with ubuntu in the end seeing as I know it better.

First issue i find is that we need to use noetic as the ROS base, but this comes with two packages not available, both ros-noetic-rosserial-server and ros-noetic-robot-upstart are missing from ROS repo. No biggy i thought, I will make them from source. Only issue is both serial-server and robot upstart conflict with an error of cannot overwrite a specific file as one or the other uses it. I think this may have been roslint or another dep of robot-upstart that caused the issue. 

After several hours of fiddling I get to a point where I try to catkin make antdroid with serial server in the src repos, thinking I can remove it from my system and include it this way (allowing me to try and install robot upload instead via make) but when i tried to  make install it failed on rosserial_windows.... 

Well i dont need that, so i thought I would remove rosserial_windows and I get slightly further, with another package in serial with issues.....

In the end I think hmmm let me even see if i can catkin make the antdroid project src by itself, nope, I get an issue with cam control trying to make.

I suppose my point is this, I do want to pick through this and try and get somewhere with it, but have you tried to do an up to date install of any of this lately?  I see the project has not had much activity, is it dead? i would love to get this running, do a remix on the actual droid, try different servos and stuff, maybe get some claws working on it and I have a thought to add a web interface with maybe a web server running on the spider so you can login and sue a web UI to control instead of the ps controller. I thought be good to control it right off a web UI.

Any thoughts on any of this? I love the project and would love to get it up and running, it just seems everything is a little old.
thanks....

smiffy6969, 2021-02-15:
So I finally managed to get i to catkin_make the antroid src by removing the camera control folder... thinking I would try and throw the rosserial package in there too, so I also cloned it into the src folder, cleaned it all up and catkin make and to see whats what.... 

CMake Error at /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "mega2560_EEPROM" links to itself.
Call Stack (most recent call first):
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:993 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:11 (generate_arduino_firmware)

CMake Error at /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "mega2560_ServoEx" links to itself.
Call Stack (most recent call first):
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:993 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:11 (generate_arduino_firmware)

CMake Error at /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "mega2560_firmware" links to itself.
Call Stack (most recent call first):
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:11 (generate_arduino_firmware)

Is what I get out....

smiffy6969, 2021-02-16:
Little update on the camera controller issue....

CMake Warning (dev) at antdroid/antdroid_cam_control/src/CMakeLists.txt:12 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "sensor_msgs_gencpp" of target "cam_control" does not
  exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

Is what shows up when you try to make with this included... i found a reference to adding the following

sudo apt install libmove-base-msgs-dev

but this did not help, found it in another thread somewhere... the result seems to be freezing at 96% make... if i ctrl + c and rerun i can get it to error....

/home/ubuntu/catkin_ws/src/antdroid/antdroid_cam_control/src/antdroid_cam_control.cpp: In member function 'cv::Mat ImageConverter::trackBall(cv::Mat, bool)':
/home/ubuntu/catkin_ws/src/antdroid/antdroid_cam_control/src/antdroid_cam_control.cpp:141:71: error: 'CV_AA' was not declared in this scope
  141 |         circle( src, Point(c[0], c[1]), c[2], Scalar(175,255,255), 3, CV_AA);
      |                                                                       ^~~~~
make[2]: *** [antdroid/antdroid_cam_control/src/CMakeFiles/cam_control.dir/build.make:63: antdroid/antdroid_cam_control/src/CMakeFiles/cam_control.dir/antdroid_cam_control.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2169: antdroid/antdroid_cam_control/src/CMakeFiles/cam_control.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed

Is this connected to the missing sensor_msgs_gencpp which might to be part of common_msgs but still trying to get to the bottom.

So i have removed the cam control again for now but again the makes are still stalling (last night i got it to complete by ctrl+c after a bit and rerun, eventually it did make)

smiffy6969, 2021-02-16:
Several runs and it now passes after i ctrl + c and rerun.... not sure why catkin_make is hanging like this

####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ubuntu/catkin_ws/build"
####
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Speed
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Walk
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Log
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Balance
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Rotate
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Gait
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Calibrate
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Foot
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_Height
[  0%] Built target _antdroid_msgs_generate_messages_check_deps_MoveLeg
[ 17%] Built target antdroid_msgs_generate_messages_nodejs
[ 36%] Built target antdroid_msgs_generate_messages_eus
[ 54%] Built target antdroid_msgs_generate_messages_lisp
[ 71%] Built target antdroid_msgs_generate_messages_cpp
[ 91%] Built target antdroid_msgs_generate_messages_py
[ 91%] Built target antdroid_msgs_gencpp
[ 91%] Built target antdroid_msgs_generate_messages
[ 92%] Building CXX object antdroid/antdroid_teleop/CMakeFiles/antdroid_teleop_joy.dir/src/antdroid_joy.cpp.o
[ 94%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/antdroid_teleop/antdroid_teleop_joy
[ 94%] Built target antdroid_teleop_joy
[ 96%] Building CXX object antdroid/antdroid_control_interpreter/src/CMakeFiles/control_interpreter.dir/control_interpreter_core.cpp.o
[ 98%] Building CXX object antdroid/antdroid_control_interpreter/src/CMakeFiles/control_interpreter.dir/main.cpp.o
[100%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/antdroid_control_interpreter/control_interpreter
[100%] Built target control_interpreter

OK.... so we have it compiled, clearly I dont have robot_upstart and cant make it from source due to the issue i raised earlier... heres the trace trying to make install

Selecting previously unselected package robot-upstart.
(Reading database ... 130170 files and directories currently installed.)
Preparing to unpack .../robot-upstart_20210216-1_arm64.deb ...
Unpacking robot-upstart (20210216-1) ...
dpkg: error processing archive /home/ubuntu/ros-deps/robot_upstart/robot-upstart_20210216-1_arm64.deb (--install):
 trying to overwrite '/usr/local/.rosinstall', which is also in package ros-noetic-rosserial-server 20210216-1
Errors were encountered while processing:
 /home/ubuntu/ros-deps/robot_upstart/robot-upstart_20210216-1_arm64.deb

For now, this only adds services on startup, I may be able to handle this another way...

smiffy6969, 2021-02-16:
Trying to do the last step in the installation instructions now, push to arduino

catkin_make antdroid_antfirm_firmware_upload || rm -rf build && catkin_make antdroid_antfirm_firmware_upload

the following error is now present (which is the same error if i try to catkin_make with rosserial in the src folder, but this time I have it installed to my ubuntu distro with a manual cmake)... so going to have to go lower  :(

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Arduino SDK version 1.0.5: /usr/share/arduino
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Generating antfirm
-- Configuring done
CMake Error at /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "mega2560_EEPROM" links to itself.
Call Stack (most recent call first):
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:993 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:11 (generate_arduino_firmware)

CMake Error at /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "mega2560_ServoEx" links to itself.
Call Stack (most recent call first):
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:993 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:11 (generate_arduino_firmware)

CMake Error at /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "mega2560_firmware" links to itself.
Call Stack (most recent call first):
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/noetic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:11 (generate_arduino_firmware)

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[3]: *** [antdroid/antdroid_antfirm/CMakeFiles/antdroid_antfirm_firmware.dir/build.make:57: antdroid/antdroid_antfirm/CMakeFiles/antdroid_antfirm_firmware] Error 1
make[2]: *** [CMakeFiles/Makefile2:2509: antdroid/antdroid_antfirm/CMakeFiles/antdroid_antfirm_firmware.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2435: antdroid/antdroid_antfirm/CMakeFiles/antdroid_antfirm_firmware_upload.dir/rule] Error 2
make: *** [Makefile:1150: antdroid_antfirm_firmware_upload] Error 2
Invoking "make antdroid_antfirm_firmware_upload -j4 -l4" failed

smiffy6969, 2021-02-16:
So.... thinking out the box, I know I can map a usb drive into a docker container.... I see the dockerfile and think lets build this in docker, and go at it that way... maybe I could add a docker-compsoe file, map the usb add in other stuff and fire it up in a container.

 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8)

I see the platform is amd64 on this, so you been using it to test compile I see on PC to test the build process... no reason why we could not dockerize this on ARM, done it before for a few things.

Anyway I will try and build the container on PC rather than PI....

paul@Study:~/Development/robot/antdroid$ docker build -t antdroid .
Sending build context to Docker daemon  10.76MB
Step 1/8 : FROM pando85/armhf-ros-indigo
latest: Pulling from pando85/armhf-ros-indigo
badc81e7e4a0: Pull complete 
e97076abb2b7: Pull complete 
9ebfaba10c07: Pull complete 
5b274878a58c: Pull complete 
b7fe3d7fbbc7: Pull complete 
4f3e66b6ac84: Pull complete 
53ba913813b8: Pull complete 
30c6a5555a34: Pull complete 
71ab50e6b0a0: Pull complete 
7f2a675b068e: Pull complete 
523aa340adb6: Pull complete 
81de69d2fba9: Pull complete 
d2c5a4b2f753: Pull complete 
895e5b62abc7: Pull complete 
50207432fc5f: Pull complete 
Digest: sha256:b6c8fe3a10c4c9977ea1e48162ae71ec1eef7c53dec3e573c0e5e2aff5376c51
Status: Downloaded newer image for pando85/armhf-ros-indigo:latest
 ---> fe843ac542ac
Step 2/8 : RUN apt-get update && apt-get install --no-install-recommends -y     ros-indigo-ros-comm ros-indigo-diagnostics ros-indigo-joystick-drivers ros-indigo-rosserial ros-indigo-rosserial-server ros-indigo-rosserial-arduino ros-indigo-robot-upstart ros-indigo-bond-core ros-indigo-dynamic-reconfigure ros-indigo-nodelet-core ros-indigo-class-loader ros-indigo-image-common ros-indigo-vision-opencv ros-indigo-image-transport-plugins ros-indigo-angles     && rm -rf /var/lib/apt/lists/*
 ---> Running in be2373e5f2ef
standard_init_linux.go:211: exec user process caused "exec format error"
The command '/bin/sh -c apt-get update && apt-get install --no-install-recommends -y     ros-indigo-ros-comm ros-indigo-diagnostics ros-indigo-joystick-drivers ros-indigo-rosserial ros-indigo-rosserial-server ros-indigo-rosserial-arduino ros-indigo-robot-upstart ros-indigo-bond-core ros-indigo-dynamic-reconfigure ros-indigo-nodelet-core ros-indigo-class-loader ros-indigo-image-common ros-indigo-vision-opencv ros-indigo-image-transport-plugins ros-indigo-angles     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1

Hmmmmm OK, so htis is not building in docker either. I would assume the update from base killed it as indigo is now way old, same problem I think just a different angle.

Im running out of ideas. I may just have to dust off my C, read a hell of a lot and start from the beginning with ROS. The only way I can see me resurrecting this is to create a brand new project and migrate in the code bit by bit solving issues... thi sis going to take a hell of a lot of time to learn the basics of ROS. This could take some serious time to get going.

smiffy6969, 2021-02-20:
HI... UPDATE...

So I played wtih ROS2 did all the tutorials and I came to the conclusion its a box of frogs. There is so much changed, not available, not compiled for this version (foxy) use of outdated libs its a mess.

SO....

Im gonna do what I do best, nerd the screw out this.

I am going to take the firmware files, pull them apart, compile and push to arduino.....

Work out the messaging over serial....

Create a node express server to output serial commands over serial to arduino...

Make a web UI using web components to design a controller interface to talk to the express server...

I have made some progress, getting the first few files compiled on arduino, got node talking to arduino over serial too.

I will dockerize this when done, make available an image to setup the servo. We should be able to do away with the larger pi and pi zero this too I would guess.

watch this space, I will resurrect this in one form or another!

pando85, 2021-02-20:
Hello @smiffy6969, I'm glad to see that you are fighting with this outdated code to bring it back to live. Good luck. If I could help you in something I will stay here reading your progress.

smiffy6969, 2021-02-22:
Thanks... Im making progress, i think treating the firmware separate from the whole ROS ecosystem may be a good thing... the big issue with building it in ROS means monolithic lock in. Hopefully a modular approach to this will benefit all. The ability to choose the method of control be it ROS/2 or any other mechanism will make it stronger.

I am quite surprised at the current state of ROS/2 I have to say, for a framework, its a great concept... I am not new to architecture design and the design feels good, the state of the ecosystem is a bit nuts I have to say. It feels harder to get it configured and setup than it does to actually use it, and thats if you can find packages that are migrated.

I think first and foremost make the firmware independent, make it lean, and then how you choose to interact is up to you, I will go with API and web based UI, as its what I know and lends some benefits... i can leverage tensor flow for clever stuff too.

But all this said I will ensure it is usable by ROS too, should anyone want to go that route.

You guys have a great thing here, I think this will be a rewrite more than a fork, but the bits that survive will have the strongest of thanks and connection to this great work! Ive only just discovered the joys of 3d printing, and I am glad i get to combine it with my love or architecture design and coding too...

thanks and I will keep in touch on this.

smiffy6969, 2021-02-22:
ITS ALIVE!

Issued first command... through a new communications layer from node to walk 10 by 10...

response:
Error: Hexapod::IsCollising == true

Its talking to me now :)

I know it seems small but, its the first proper response back with new bootstrapping and com layer.

I need to look over the code imported, I may just leave that as is not sure yet.

So I think next job is to throw up a quick API and UI layer to issue commands and dump logging back. Then we can attempt to connect some servos and try a calibration....

This is huge progress!

smiffy6969, 2021-02-24:
So I have generated a new project... an api from my other project cerberus mvc... I have just updated it to work with socket.io so I can have a split API on the PI.

The idea is to have login to the PI over a UI... so we log in to the PI over REST, basic things done over rest API...

Then when we connect to roamer, we will do this via web sockets. This should allow for stopping it if we loose comms, real tiem updates to the UI over sockets and better control.

I have just updated the MVC stack to work with sockets and tested it out and am now going to look at building a UI in my other project custom-web-component.net.

We should be able to hit the pi in the roamer from from anywhere, wake it up, connect and drive it from where ever you want, once camera feedback sorted.

I have some work to do on connecting the servos up and testing calibration and building the UI part but this is not too difficult, just time consuming. 

So I have a plan now, it should all be doable now the firmware is compiling, fingers crossed.

smiffy6969, 2021-02-27:
Hi,

So I have my basic API server talking to the arduino over serial.... im getting help messages back now from my new Com c++ file....

I have put together a quick script to send calibration values in and I can see they are being save when i power down and up and read servo data from eeprom I can see the values are sticking.

The issue I have is that I cannot seem to get a single servo to move on any digital IO pin. I started at 22 na dmoved the servo (just one) down the header 22, 23, 24 and so on.... im sending the calibration message 253 to ensure move is on, then the calibration value say 50, then the calibration save 255. Nothing, I dont see any servo movement and the servo appears to not be in any torque (i.e. I can move it).

Tried all the digital pins and also tried scripts to run through all configurations of legs and members with the above hoping to see a movement on the servo. But nothing.

I have power... I have checked servo voltage. I have also boosted the voltage to 5.8 with a seperate battery. still nothing on the servo in any of the digital IO pins from any of the calibration legs or member positions.

Example of messages through my new com file (shorthand messages) are...

where 

calibration [leg] [member] [number]

c 5 2 255
c 5 2 22
c 5 2 253

giving the response from arduino...

Info: You must restart your Arduino now
Info: Calibration completed
Info: Ang [252]: servos won't move to the calibration value
Info: Ang [253]: servos will move to the calibration value
Info: Ang [254]: shows current calibration.
Info: Ang [255]: validates FIRST calibration.
Info: Moving servo to default position
Warn: Save completed.
Warn: Writing to EEPROM memory. Do not turn off the power source
Info: Ang [252]: servos won't move to the calibration value
Info: Ang [253]: servos will move to the calibration value
Info: Ang [254]: shows current calibration.
Info: Ang [255]: validates FIRST calibration.
Warn: Move and calibrate ON
Info: Ang [252]: servos won't move to the calibration value
Info: Ang [253]: servos will move to the calibration value
Info: Ang [254]: shows current calibration.
Info: Ang [255]: validates FIRST calibration.

Should I not see the servo trying to move to position at any point, or at least being in torque... i.e. cannot move?

Any thoughts?

smiffy6969, 2021-02-27:
OK i got this... i worked out how to toggle pins on IO and they appear to be working....

I went back, had missed out the hexbot start (com start was there) and that led me on to the pin maps in teh configuration file... choose pin 52, calibrated and off it went, servo now moving.

getting closer lol.

smiffy6969, 2021-02-28:
I am managing to calibrate the servos now with defaults.... but I am attempting a move now and just get the collision warning. Some dumps show

Error: (CoxaOffsetX + FootDistance * GetCos(LeftFrontCoxaDefaultAngle)) / 2 = -1685
Error: GetCos(LeftFrontCoxaDefaultAngle) = 7071
Error: FootDistance = 120
Error: CoxaOffsetX = 78
Error: _position[0][0] = 0
Error: max_distance = 0
Error: Hexapod::IsCollising == true

that top number looks really high! thats the check to see if it will collide right? Is this because I needs to set a gait? does it think the legs are folded in?

I suppose now the question is what are the normal steps you would run through to get to a position where you can move the bot? 

Would you calibrate, set gait, do something else, then move is capable? something like that. Its the operating of the bot I am struggling with a bit now I guess.

smiffy6969, 2021-02-28:
Further attempts to set the foot distance higher to get past collision detection makes servos move on a move/walk of say x 1 y 1 but then i get angle errors...

Error: Tibia = 128
Error: Femur = -12
Error: Coxa = 90
Error: Leg: = 4
Error: Angles are not in range

which would be calibration wrong?

I am attempting to just get servos moving from commands for now, on a bench, no bot, just the arduino and a few servos connected so i can seee them twitching but the whole calibration and general usage is escaping me a little.

smiffy6969, 2021-02-28:
So I worked out your leg order is left right left right left right.... 

I have now given the thing a calibration of default values from your wiki to match the numbering...

It is doing a sequence on the legs now, moving with x 1 and y 1 is fine... 2 is fine x 3 y 3 or any number over three appears to do part of the sequence but then stop, no other commands work after that. I can see the loop appears to stop in the ino file that kicks the library files.

Im thinking power right... so I hook up several battery packs, remove all servos but one.... still same issue
I can set logging to debug.... then send move 100 100 and it does the steps slow, spitting out responses, and will repeat once its finished fine. This is slower for sure, and one servo at a time so maybe it is power thats the issue. I am running a 5v duracel portable power pack that gives out up to 2.5amp @5v max so it should not be a power issue, maybe the dip is affecting the arduino and crashing it.

Im a bit stuck on this now to be honest, not sure if this is due to calibration or just the setup. The 5v pack and a 4 cell aaa pack on teh arduino with one servo should be more than enough power (including 400ma ish from usb) so I am thinking its more likely the firmware.

any ideas would be most appreciated.

smiffy6969, 2021-03-15:
Hi,

Not been here in while. whats been happening?

Good news bad news. I have decided to ditch the raising of this project due to the amount of issues and also some of the dependencies used.

That's the bad, i will not be raising this project. 

Good News, I still want this to be a thing, so I have started building from scratch.

I currently have the prototype code at a stage where we can simulate a walk, step, stand, sit and other basic commands. I have a base level API that sits on PI, using serial to communicate with the arduino. I am currently creating a socket based UI to give control via the API.

Took a while to get the arduino code right, but there is now a real simple servo chaining library I wrote to allow connection of N servos... as a chain. You use kinematics to turn movement to angles in a sequencer, which adds all leg parts to a chain, the chain can use delay and time to orchastrate movement on many many servos using the base Servo class in arduino.

This work very very well. building sequencies is very very simple. The orcha class then takes care of the reset moving all servos in the chain the amount need to get them to finish within there time and after there delay.

This method allows me to chain all servos and perform a walk in just 3 groups within the step sequence. This also allows us to throw sequences out super quick.

Currently supporting sit, stand, tiptoe, walk, crab postures with crabforward, back left right, stepforward, back left right, crabforward, back left right all simulated on one half of the bot (i only have half the legs at present) and other sequences planned, all driven through a test UI, sockets and a socket/rest based API on the PI. 

Once I am a little further I will be releasing the orcha library as a stand alone library for arduino.

I have all this working at present with only 5% storage and 8% memory on the mega, with most of the prototyping done on the uno (needs more timers and IO, but would be fine for 4 legs).

I also have thoughts on making this 8 legs when my printer is up and running again with pan on camera.

All this will be working over internet connections, so the aim is to be able to drive it from home or the other side of the world should you wish, maybe a docking station might be a thing, wireless charging may be interesting.

So very much alive, I just felt there was too much needed to happen. currently passed r and d phase now with Ui being planned whilst waiting to setup printer again. As soon as I have more times on when this could be ready in V1 along with code (will be MIT) I will post.

Thanks for the inspiration.

jftheoret, 2021-04-19:
Thank you very much for trying to make this work! It is a great design and would love to see it up and running. Looking forward to see your work and maybe trying to contribute!!

smiffy6969, 2021-04-20:
Thanks. I have made some considerable progress, once with re engineering this codebase, but then finally with re-writing all of it again from scratch. I wasnt happy with using both arduino and pie, i thought it was overhead so i went purely with pi and a HAT.

To be specific the new bot, now called ROAMer runs an a raspberry pi zero with a PWM HAT capable of stacking and runnning 900+ servos over i2c bus.

This means way less power, about 30% the size physically, proper PWM on chip timing and one single codebase to handle API for wireless interaction and bot kernel.

It now purely runs on node with a python bridge to i2c. All servos use my own orchestration class allowing us to move and plan sequences of moves using kinematics to calculate angles. Sequences are simple classes we can now drop in. I have done some basic ones and they can use time and delays to control complex movements in parallel.

You can find the project @ ulsmith on git hub which you can find through my account on git hub (most projects go under ulsmith organisation).

So where are we at? the UI is now going to be a PWA installable web app with login to ROAMer from where ever you like.

I am currnetly in the process of moving house so all dev has paused, once i am moved i will pick back up but your welcome to look through the codebase, its functional and working, we can sit, stand crab, walk run at present.

My aim is to make this simpler, lighter, faster and more maintainable with less chance of becoming a paperweight lol.

willietes, 2023-06-08:
@smiffy6969 I started playing with this project and have made some significant progress.
I have succesfully compiled all the code and even uploaded the firmware to the controller and was able to send through calls to a servo! I am using ubuntu 18 and melodic code base. I was able to compile the code on my Pie with some tweaks and now figured out what I did wrong on the pie with uploading firmware.

Anyone feel free to ask. I am even open to exporting an OVA of my working ubuntu instance.